OMOTE-Community/OMOTE-Firmware #1: Add gerber files

thijstriemstra, 2023-07-06:
Would be great to have the gerber/bom/placement files so it can be produced by a company like JLPCB or PCBWay.

CoretechR, 2023-07-06:
Hi, you're right, I will add those files to the repository soon.
Until then there are plugins for both JLCPCB and PCBWay in KiCad that work really well.

mandreko, 2023-07-07:
I just generated these in KiCad in a way that has worked for me for previous PCBs, if it's helpful.

[OMOTE_gerber_20230707.zip](https://github.com/CoretechR/OMOTE/files/11985526/OMOTE_gerber_20230707.zip)

thijstriemstra, 2023-07-07:
thanks @mandreko. I believe we'll also need the bom and placement csv files for jlpcb (?).

CoretechR, 2023-07-08:
Added the GERBER files (7b5d897)
These were exported by the JLCPCB plugin, but should work for other manufacturers as well.

closed 2023-07-08 by CoretechR