OMOTE-Community/OMOTE-Firmware #3: BOM and sourcing components

mandreko, 2023-07-07:
Firstly, I love this project. Not just from it's sole purpose, but also for someone trying to learn electronic engineering better. I am studying the PCB to try to figure out how everything was done and the purpose of each yet.

I figured I'd start to build at least one of these. I created a BOM here: https://octopart.com/bom-tool/xXZMKVQA

The issues I have thus far, are:

1. Making sure everything on that BOM is correct. 
2. Sourcing specific components that seem to be obsolete or not readily available.

Can you check that this BOM is correct, or supply one separately, to address issue 1?

For 2, I'm finding that SFH4346 is considered "obsolete" and not readily available anywhere that I can find. I think that SFH4546 might be a replacement, but I'm unsure. Additionally, trying to source the CH340C is tough as far as I can tell outside of LCSC. Is that where you got yours, or did you find it (or an alternative) elsewhere?

Thanks!

thenicnet, 2023-07-07:
Just a heads up, the **MIKROE-1120** is US Only from Mouser and DigiKey.
Anyone outside the US might have to source a comparable battery.

mandreko, 2023-07-08:
That's definitely a good note. I didn't specifically look for availability per country, but it's likely wise. I imagine there's a similar battery for other countries, but we may have to search for it some more.

CoretechR, 2023-07-08:
Hi @mandreko , I've added a BOM to the project now:
https://github.com/CoretechR/OMOTE/blob/main/PCB/BOM.csv
These are the parts that I ordered when I built the remote. Maybe you can check that against your list.
The IR LED should not be an issue. You can probably use any 3mm IR LED.

retorquere, 2023-07-09:
If anyone in Europe is looking to sell an assembled one, I'm interested.

syntaxerr66, 2023-07-12:
For those trying to source the CH340C, I just ordered some from this aliexpress vendor. Fingers crossed they're legit and work, but it was only $8 total, so it's worth a shot.

https://www.aliexpress.us/item/3256804957277020.html?gps-id=platformRecommendH5&scm=1007.14452.226711.0&scm_id=1007.14452.226711.0&scm-url=1007.14452.226711.0&pvid=708ccaa0-5d3b-4b62-a6b7-d671f99858c8&_t=gps-id%3AplatformRecommendH5%2Cscm-url%3A1007.14452.226711.0%2Cpvid%3A708ccaa0-5d3b-4b62-a6b7-d671f99858c8%2Ctpp_buckets%3A668%232846%238113%231998&pdp_npi=3%40dis%21USD%212.77%212.77%21%21%21%21%21%402101f49416778227342383096ecfbe%2112000031838125062%21rec%21CA%21&ad_pvid=202303022152142686206942585875142559_2&gatewayAdapt=glo2usa4itemAdapt

bjoerns1983, 2023-07-13:
If anyone would sell assembled PCB's i would be interested. 3D Printing is not a problem.
Love the Idea of an remote you could also use for MQTT!

moserja, 2023-07-14:
Did R1,R38,R39,R40 get removed? I don't see them in the BOM.csv file.

CoretechR, 2023-07-15:
> Did R1,R38,R39,R40 get removed? I don't see them in the BOM.csv file.

Good that you noticed, R37 was missing as well. Should be fixed now.

dekiesel, 2023-07-15:
Is there any way to subscribe to see when somebody decided to sell a ready made pcb (in the EU)? I'd love to have one of those, but my attempts at smt soldering have been very fruitless so far.

CoretechR, 2023-07-15:
@dekiesel If something comes up, I will definitely post it on Hackaday.io and Twitter.

JediMongoose, 2023-07-16:
I'd also be very interested in buying an assembled PCB.. had a look on tindie.com, but nothing yet :)

mandreko, 2023-07-16:
@JediMongoose Give it a couple weeks, and there likely will. I’ll be at least buying components for enough boards to put up there in small quantities.

JediMongoose, 2023-07-16:
> @JediMongoose Give it a couple weeks, and there likely will. I’ll be at least buying components for enough boards to put up there in small quantities. (US based)

I ended up here due to the hackaday feature (https://hackaday.com/2023/07/15/hackaday-prize-2023-omote-universal-remote/) - I'm sure there'll be more :smile:

hypercube33, 2023-07-18:
I'm doing a struggle to see if I can get PCB's made and see how impossible this is to solder myself. I've never done more than through-hole so this should be interesting. I have an immediate need for at least 2 of these, ideally expanding them with a hub when that is finalized. Also would love to help in any way! My life is a void now that Logi dropped Harmony.

mandreko, 2023-07-18:
If it’s helpful, I’ve purchased the PCBs and components for about 10x remotes. Once they come in, I’m more than happy to build some for others. I enjoy soldering.

ramonflores08, 2023-07-18:
I would do the same but don't have the funds to buy so many 😕

Dark1886, 2023-07-20:
I will also have a few boards that I will have left over, but they will not have the screen or battery installed. Those are easy additions though.

ramonflores08, 2023-07-20:
I will have 4 pcbs left over from the batch I ordered.if someone wants one and is willing to buy the components I can solder them.

Wolverine-14, 2023-07-21:
> I will have 4 pcbs left over from the batch I ordered. if someone wants one and is willing to buy the components I can solder them.

I would be interested.

tnph, 2023-07-24:
> If it’s helpful, I’ve purchased the PCBs and components for about 10x remotes. Once they come in, I’m more than happy to build some for others. I enjoy soldering.

I'm interested in buying a populated PCB, and willing to pay in advance if required

dekiesel, 2023-07-24:
It would be helpful if the people that are offering to help others would say where they are located.

GaryTill, 2023-07-24:
I'm also in Australia if anyone is selling partially or fully assembled PCBs.

demartinoj, 2023-07-25:
> I will have 4 pcbs left over from the batch I ordered.if someone wants one and is willing to buy the components I can solder them.
> 
> 
> 
> I'm from the US

Hey! I'm located in the US and am happy to buy a pcb in whatever state it is currently in so no additional work for you. I'm happy to source and populate whatever components aren't already there. 

Thanks so much! Feel free to dm me! 

Amazing project by the way!!!

bjoerns1983, 2023-07-25:
Anyone has already ordered the PCB from JLCPCB?
Are there any special settings to be made, or just trhowing in the gerber.zip for PCB and Stencil and you are done?

ramonflores08, 2023-07-25:
I ordered them from jlcpcb, I should get them today.  I will let you know how it goes. I didn't add any special settings, and got the stencil as well as the pcbs.

bjoerns1983, 2023-07-25:
Great, looking forward to hear from you

mandreko, 2023-07-25:
> Anyone has already ordered the PCB from JLCPCB?
> Are there any special settings to be made, or just trhowing in the gerber.zip for PCB and Stencil and you are done?

I did nothing special. I just have them the gerber file and they look perfect. I’ll be soldering them once my last component arrives soon. 
[image]

bjoerns1983, 2023-07-25:
Looks good.
Are there any services where you can order the other parts supplying them the BOM or  did you order every parts seperately?
Maybe letting JLCPCB solder at least one side would also be an idea.

mandreko, 2023-07-25:
I ordered from Newark, Mouser, Digikey, and one part from AliExpress (the one I’m waiting for). I used octopart to identify the cheapest part locations. I used this BOM that I made from the official BOM: https://octopart.com/bom-tool/xXZMKVQA

Dark1886, 2023-07-25:
Single sided soldering at JLCPCB could be pretty good guidance for most since it allows for many of the small components to be added and only requiring the buttons to be soldered on later. Although I noticed when I entered my parts into JLCPCB that many of the parts didn't have the correct orientation or placement, so you need to be very careful before ordering.

GaryTill, 2023-07-26:
PCBWay are telling me it can be done, double sided assembly, they can source parts if needed, very good price and a decent discount for first time customers, but gotta get 5 done as a minimum I believe.

bjoerns1983, 2023-07-26:
Wasnt PCBWay also the guys with the shared project feature?

Dark1886, 2023-07-26:
@bjoerns1983 You can have JLPCB source and solder the components to the board for you, either single or double sided. If the board width is too thin they will add additional breakaway tabs to the side to meet the minimum spec.

GaryTill, 2023-07-27:
I decided to give it a shot myself rather than getting it pre-assembled and have ordered 10 boards a paste stencil and components to build up a couple of units.

Haven't done a more complex board like this myself for about 10 years so I have to buy some new gear and drink less coffee but after looking over the board it looks like just a small hotplate will do and it shouldn't be too hard.

mandreko, 2023-07-27:
I also bought separate components to diy. I’ll have to check the price breakdown but I feel like it’s much more expensive to diy for some reason.

Shad0wPhax, 2023-07-30:
> I will have 4 pcbs left over from the batch I ordered.if someone wants one and is willing to buy the components I can solder them.
> 
> 
> 
> I'm from the US

I'm interested, I'm in the US as well. Let me know how how to get contact you. 

Thanks!

tnph, 2023-07-31:
I have a similar experience to Dark1886 trying to order a fully populated PCB from JLCPCB, several components incorrectly positioned/orientated. eg J2 TFT header was 180 degrees rotated and a long way from the solder pads.  Q1,3,4,5 diodes were 180 rotated, also some capacitors seem to be 180 rotated (I'm assuming the SMT ones have a polarity?). 

I was wondering if anyone had successfully ordered from JLCPCB and had an order they could share?

Example of Q4 180 rotated:  
[image]

gacekk, 2023-07-31:
> I will also have a few boards that I will have left over, but they will not have the screen or battery installed. Those are easy additions though.

Interested in EU

Dark1886, 2023-07-31:
> I have a similar experience to Dark1886 trying to order a fully populated PCB from JLCPCB, several components incorrectly positioned/orientated. eg J2 TFT header was 180 degrees rotated and a long [...]

I did just receive my boards from JLCPCB, I can try to share my order if that is an option on the site. I had to check orientations and position of all components just to make sure it came back as expected.

mandreko, 2023-07-31:
If it's helpful to anyone, I had PCBWay do a quote for 20x boards with components and assembly (minus 2 parts I couldn't find in their database). It was $1,698.31 USD for the 20x, or roughly $85 per board. My previous statement about DIY being more expensive is incorrect now with their updated quote.

GaryTill, 2023-08-01:
Got a package of 10 PCBs and components to get started on a couple of units today. Probably wont be actually assembling for a few weeks but I will have some bare boards leftover if anyone in Australia wants some for the cost of shipping.

Dark1886, 2023-08-01:
@mandreko Have you built up one of these boards yet? I noticed that the connector on the MIKROE-1120 battery does not actually match what the bom calls out for the on-board battery connector. The part on the board will need to be changed or a connector modification may be needed to use correctly.

mandreko, 2023-08-02:
> @mandreko Have you built up one of these boards yet? I noticed that the connector on the MIKROE-1120 battery does not actually match what the bom calls out for the on-board battery connector. The part on the board will need to be changed or a connector modification may be needed to use correctly.

I have not yet. I'm still waiting on my CH340C to arrive :(

I did go through my box of components, and you're absolutely right. I completely missed the connector type! The MIKROE-1120 uses the JST-XHP-2 connector, and our OMOTE connector needs a JST-PHR-2 connector. Those definitely won't match.

I suspect that the *easiest* things would be:

1. Return the battery and order a new one with JST-PHR-2
2. Modify the battery's connector to be a JST-PHR-2 with a crimping set

I can't really return my current batteries, so I guess I'll be going with option 2. I luckily have some PH2 connectors here that should hopefully work. If you *are* able to return yours, I think one of these may be an option for US:

- https://www.amazon.com/2000mAh-Battery-Rechargeable-Lithium-Connector/dp/B07S75L945
- https://www.amazon.com/2000mAh-123545-Lithium-Rechargeable-Connector/dp/B0C2KMHD5H

For outside the US, you may have to search in your local retailers.

ramonflores08, 2023-08-02:
I just soldered the battery to the pads directly. Guess that's option 3

tnph, 2023-08-03:
Just thought people might be interested to know I just ordered 5 fully populated boards (minus batteries and TFT displays) from PCBWAY and it came to $228.58 AUD which is $145.71 USD.  I got a $30 discount for my first order and another $5 discount voucher which randomly popped-up! Seems reasonable...  

I too am concerned re component position and orientation so asked them to confirm with me before assembly, and they replied with "We will match component with BOM and board before production. And send you sample picture for confirmation before production".. so I'm a little more confident now..!

MikRED97, 2023-08-05:
> I will also have a few boards that I will have left over, but they will not have the screen or battery installed. Those are easy additions though.

@Dark1886 If you have any preassembled boards available shoot me an email. 
Otherwise if someone in the US has any or is going to put some boards together I'd be interested in one.

mandreko, 2023-08-08:
I've also now found that the USB ports in my Octopart BOM are incorrect. Without waiting a long time for China to ship, I found that the specific USB adapter (TYPE-C-31-M-12) appears to be used by mechanical keyboard enthusiasts and I grabbed a few from here: https://keeb.io/products/usb-c-port-12-pin-hro-type-c-31-m-12

The pricing is likely still better if you go from a major distributor, but for US-based shipping, that was the quickest way I could find last night.

ramonflores08, 2023-08-08:
I got the USB port from Mouser. Not much cheaper, but might help.

https://www.mouser.com/ProductDetail/640-USB4105-GF-A

tnph, 2023-08-09:
I really want this project to be a success!
On order are 5 fully populated PCBs from PCBWAY "build time 23-25 days"..
So far all in my possession is one TFT screen and one LiPo battery from (Little Bird Electronics Australia) 
...and I'm already excited! 

Hoping to replace these :-)   (at least the major functions)

[image]

GaryTill, 2023-08-09:
I'd have a similar number of remotes per room if I hadn't already combined them all into one hacked OneForAll remote (surprisingly powerful if you get a programming cable). My issue is keeping the active device on the remote synched with my IP controlled video switching setup which has 8 sources and 8 zones, then adding to that one theatre zone with a few local devices in addition to the 8 coming from the switcher.

Selecting Movie mode on the touch panel then sitting in the couch is all good until someone complains the remote isn't working because it's still set to the device mode for TVTuner2 when they're watching Shield3...

grover31, 2023-08-11:
> Just thought people might be interested to know I just ordered 5 fully populated boards (minus batteries and TFT displays) from PCBWAY and it came to $228.58 AUD which is $145.71 USD. I got a $30 [...]

I would be very interested if you are up for re-selling any of these (or sharing your PCBWAY config) in Tas Aus.

tnph, 2023-08-11:
> I would be very interested if you are up for re-selling any of these (or sharing your PCBWAY config) in Tas Aus.

I does look like I can share PCBWAY config/order details... but it looks like the "sharing" should be done by the creator CoreTechR as there are benefits: "_PCBWay will donate 10% the cost of PCB to you, for your contribution to the Open Source Community._" which I certainly don't deserve to be getting!   Also can share the final BOM spreadsheet which includes a few substitutions that I agreed with PCBWAY, the other files I sent them (eg. gerber) were from github.

I may have a spare PCB too depending how many I break in the process!  ETA is still 3 weeks..
[T-B2W671156A-5sets-BOM(2023-08-01).xls](https://github.com/CoretechR/OMOTE/files/12318902/T-B2W671156A-5sets-BOM.2023-08-01.xls)

tnph, 2023-08-15:
I need a bit of help from @CoretechR and anyone else who can help with confirming the orientation of some components please. PCBWAY have done initial assembly and sent some pics, as requested, with questions on the orientation of: C13, C14, IR LED, AE1 or AE0?, D2, U6.  (There's a bit of confusion over AE0 v AE1, the BOM says AE1, their pic file mentions AE0).

Also they comment: _"For C13 C14，the positions according to gerber and centroid file are exchanged,we've soldered as per Centroid file, is it correct ?"_

Apologies for the PCB colour - my wife chose it!! ...but it does make the components easy to see :-) 

I've also been doing extensive 3D print "testing" (read trial and error!) to get a good result, maybe I'll start a new thread on that if people are interested. The main tip is to print the top face-down on a clean glass bed to give a nice shiny finish! 

 
[T-B2W671156A.zip](https://github.com/CoretechR/OMOTE/files/12353569/T-B2W671156A.zip)

CoretechR, 2023-08-16:
Hi @tnph what a nice color! 

- U6: That looks correct, the IR sensor should point sideways
- D2: orientation is correct. I didn't expect them to solder the LED, wow. The LED should ideally have ca. 1mm of offset to the PCB edge. In the picture it almost looks like its bent upward.
- C13 and C14: Not sure what they mean, but the parts are in the right places
- D5: correct
- AE1: correct

Maybe someone else can double check but most of these can be figured out from the Kicad files.

tnph, 2023-08-16:
> Hi @tnph what a nice color!
>
> * U6: That looks correct, the IR sensor should point sideways
> * D2: orientation is correct. I didn't expect them to solder the LED, wow. The LED should ideally have ca. [...]

Thanks for checking @CoretechR.  Feedback from others also welcome - I'm dying to give PCBWAY the go-ahead to build all 5 boards!
The PCB is yellow 'cus I had this daft idea of printing a clear case to show off the internals - my research tells me this is very difficult to do in practice...

tnph, 2023-08-16:
Hi @CoretechR  - PCBWAY have come back again to question C13 and C14 positions, now I realise why - if you look carefully in the pic you can just make out the writing between the solder pads is opposite to the silkscreen C13 and C14, so I want to double check.

Looking at the PCBWAY pic, C13 is the physically smaller cap, when compared to C21 (another 10uF cap in BOM); also C14 and C22 are 100nF in BOM but different physical sizes in the picture.

On the face of it C13 & C14 do appear to be swapped - maybe just a BOM error? 

Kicad grab:
[image]

PCBWAY pic:
[image]

CoretechR, 2023-08-16:
@thenicnet Oh, I must have swapped the silkscreen marking for C13 and C14 by mistake. Sorry about that! 
Everything else seems to be in line with the schematic. The BOM should be correct as well. You can tell PCBWay to ignore the silkscreen marking.

M4RiO-priv, 2023-09-04:
Any success stories to share ? I am really looking forward to build that to replace my collection of remotes

tnph, 2023-09-05:
> Any success stories to share ? I am really looking forward to build that to replace my collection of remotes
I've had some success, I got PCBWAY to make 5 fully populated PCB's and after some back-and-forth on components (I had to specify substitute switches and a few other things) and placement (eg. C13/C14 mentioned above) they turned out perfect.  I was able to flash the "main" branch of code using VSCode (with PlatformIO extension). 

I've made a few basic changes to the code, like adding a button to publish an MQTT message, which is picked-up by Home Assistant to perform an automation. This was a simple copy/paste/change of existing code as I'm not a C++ dev!  I found a VSCode plugin VSMqtt which worked well for monitoring the MQTT messages.

I'm obviously keen to see the new "abstraction" version of software too as there's no way I can code anything complex!

Another thing I bought was a Broadlink RM4 Pro universal remote (AUD$60 Amazon) which can be used to learn both IR and RF codes (I used the Home Assistant integration to capture the codes, which conveniently stores them in a file). So I'll be spending some time capturing the IR codes from all my remotes in readiness for loading into Omote.

I've also spent a disproportionate amount of time 3D printing all the case parts!!  

[image]

[image]

dekiesel, 2023-09-05:
> Anyone found reliable place in EU to make those PCBs?

PCBWay sends to the EU. I haven't ordered this pcb from them, but others, and always without issues.

[63 more replies]

closed 2025-03-05 by mandreko