OMOTE-Community/OMOTE-Firmware #10: IR learning abilities

Jonas-fr, 2023-07-16:
Would you consider adding an IR-receiver to allow IR-learning abilities (via a future firmware update) to this remote? It will vastly expand the use-cases for it.

Jonas-fr, 2023-07-17:
This modular approach is sound! Do you have more details about such an external IR receiver and where it could be plugged in on the current remote?

riker09, 2023-07-17:
Not sure if it can be helpful here, but there is Flipper, a hacker tool similar to this project that allows reading IR codes (and replay them).

CoretechR, 2023-07-17:
There's an IR receiver built into the remote already (U6). Learning codes is currently not used in the software, but I have tested that the sensor works.

[image]

Jonas-fr, 2023-07-17:
> Not sure if it can be helpful here, but there is [Flipper Zero](https://flipperzero.one/), a hacker tool similar to this project that allows reading IR codes (and replay them).

I know very well about it but it's too expensive just for one use case (IR learning).

> There's an IR receiver built into the remote already (U6). Learning codes is currently not used in the software, but I have tested that the sensor works.
> 
> [image]

My bad, I didn't saw it along with IR LEDS. Well this is perfect ! Would you be open to future PR regarding IR learning functions then or do you want to focus on the TX side for now?

CoretechR, 2023-07-17:
I can't promise anything right now. 
Honestly I wanted to take a break from this project as I have put too much time into it already. 
But since so many people seem interested, I might have to keep working on it. 

For my previous [OLED keyboard project](https://github.com/CoretechR/Keybon), I created a windows UI for configuring the screens. I imagine something similar could be done with the remote. A web interface hosted on the ESP32 would make it cross-platform, but I'm not sure about the performance. The IR learning would be part of that interface.

Jonas-fr, 2023-07-17:
You don't have to promise anything, you can (should!) take a break whenever you like! I was just asking if that (RX abilities) would be a direction you'd like for this project in the short/middle term or not.

Anyway thank you for all the work you've already put into this project!

Dark1886, 2023-07-19:
Just adding an idea that could help with IR command learning, could the commands be added into the eeprom locations to allow dynamic changing without needing to hardcode into the values in the source code. I'm hoping to work on this when I can get my hands on the parts.

CoretechR, 2023-07-19:
@Dark1886 Sure, the flash memory of the ESP32 would be the best place to store the commands. You're welcome to try an implement this.

Dark1886, 2023-08-02:
Just a thought, If adding IR learning functionality on-board may be difficult, an IR reading mode could be implemented that could spit out the received IR signal to some serial/http Post/MQTT log. It doesn't directly learn the function, but it could make capturing and programming the IR codes a lot more simple.

KlausMu, 2024-01-17:
I implemented an IR receiver, looks like this:
[image] [image]

Even recognizes the codes which the OMOTE is sending by itself :-)

I can publish this later. But currently I'm still working on some other features.
I'm afraid my IR receiver code can not be provided as a pull request to the original source file. I did more or less a complete rewrite.
I used a more modular approach, with separation of code into devices (describing their abilities), the gui parts (single files for each device), the user interfaces (keypad, touch), and a command pattern for sending and receiving commands like INCREASE_VOLUME, beside some other stuff.
Overall more than 40 (small) files. But (at least to me) much better understandable and customizable than a single file with >1000 lines.
Still need some time to finish it. So stay tuned ...

closed 2025-04-06 by KlausMu