OMOTE-Community/OMOTE-Firmware #11: hub option

TomW1605, 2023-07-17:
someone sent me a link to this project and it looks really cool, i was considering doing it myself when my harmony remote dies but i hope i wont need to.

one thing though, it mentiones not needing a hub as a good thing but i think there a quite a few functions that a hub provides that make having one a good thing. in particular it allows for your av devices to be controlled through an app, webpage or automation system wihtout worrying about where the remote is and if its pointed at the devices. it also allows the remote to control devices that may be hidden without needing an IR repeater setup.

while i do think not *needing* one is good it would be great to have the option of adding one. i dont know if it would be best to try and add a direct RF connection to a hub or just rely on them both connecting to wifi but i think having a hub of some form is great.

same goes for a dock, having contacts exposed on the case and a dock that just has 5v on a pair of pogo pins would be a great addition to eliminate the need to activly charge the remote at all

either way i love this project and will be following it with great intrest.

Dark1886, 2023-07-17:
I think having a hub feature may be nice, but I suspect that would be a bit down the road since I think there are other more important features currently needed. Although I think a hub could be used if you essentially took main MCU and used that as a hub with only a basic "client" remote that is just guiding the main hub. 

As usual with open source projects, feel free to take a stab at it since that is what will generally push the 'experiment' closer to a real feature in the project.

TomW1605, 2023-07-18:
i agree there are more important things, i really just wanted to raise it so it wasnt outright discounted as an option.

i think the hub as main with a client remote (that probably just has RF and maybe IR) is how the harmony hub bassed remotes worked but in this case i think having it as an addon so it works with the OMOTE without changing the OMOTE would work better since it means you can save that cost if you dont need it.

ill take a stab at a pcb design at some point and report back if i get anywhere

CoretechR, 2023-07-18:
@TomW1605 A hub as an optional add-on is a nice idea. Many smart remotes completely rely on a hub for communication and power, so I wanted to set my remote apart from that. But there are obvious advantages, like better IR range without having to point the remote at the device you want to control. If you want to try and make a ESP32 hub, that would be great, not just for this project alone.

hypercube33, 2023-07-18:
+1 on the hub. I agree that a hub piece for this is something I was hoping for. The project is absolutely awesome, and I intend to try to figure out how to make it since I depend on my Harmony Remote + Hub daily on all our media systems. The hub could be an ESP8266 or ESP32 running an IR blaster that the remote can send commands to output. I think there are other projects that do serial over wifi to ESP's - why this matters is because it integrates nicely with home assistant and other automations and the remote may not have line of sight to some devices which makes using it for IR annoying - the hub we have sits in a cabinet closed away but still controls IR devices nicely.

Dark1886, 2023-07-18:
@hypercube33 If you plan to design some sort of hub, it could be possible to just use the Bluetooth LE or just the LAN to connect the two, then you wouldn't require a direct line-of-sight to the hub itself.

GaryTill, 2023-07-26:
Any of the dozens of existing generic ESP based IR hubs can be easily loaded with pre-made firmware from ESPhome or Tasmota to subscribe to MQTT and blast out IR very simply.

Then it's just a matter of having some functions on the remote mapped to MQTT on a per device/room/etc basis. The IR commands could be hard coded into the remote, or it could be handled externally by nodered or similar for flexibility and smarter handling the commands without having to hardcode everything.

I mostly use direct IR in my current setup (I already have an IR distribution rig as part of the video matrix setup), I do use broadlink emitters for automation of air conditioning and such, so they can be useful, particularly for whole home control from one remote.

TomW1605, 2023-07-26:
yes hubs like that would work but i would probably want to write new code that more directly interacts with the remote.

also i think if we were to design one from scratch we could/shoud add 3.5mm jacks to have remote ir leds like many ir repeater systems use. that way you wouldent need to try and put the hub where it can see all the devices or need a seprate repeater system

Dark1886, 2023-08-07:
I think the hub will be a good idea, But I assume that will not be a main priority since the software will likely be in flux for a while, so adding additional complexity will not be priority.

closed 2025-04-06 by KlausMu