OMOTE-Community/OMOTE-Firmware #14: Can't build simulator

MatthewColvin, 2023-07-23:
Hey @CoretechR I am super interested in contributing to this project. But currently when I open the simulator project it will not build.

I was wondering if you considered forking the simulator repo and then adding it to this project as a submodule? 

I could also help set up some GitHub actions to get build validation.

CoretechR, 2023-07-23:
Hi @MatthewColvin, did you set the target in Visual Studio to x64? It might [be set to ARM64](https://github.com/lvgl/lv_port_pc_visual_studio#known-issues) by default. 
I'm not very proficient with GitHub so I just copy-pasted the simulator project in there. A proper fork might be better, I will look into it.
And thank you, any contributions are welcome! Do build validations work with embedded projects like this?

MatthewColvin, 2023-07-23:
I am set to target the x64 build. And yeah we can set up build validation for the platformio project along with the simulator. 

I am unsure how those interact with one another at this point but I look forward to digging in.

Let me get the compiler error and paste it in here

MatthewColvin, 2023-07-23:
Error	C1083	Cannot open source file: 'C:\Users\mcolv\Desktop\OMOTE\LVGL Simulator\LvglPlatform\lvgl\examples\widgets\obj\lv_example_obj_2.c': No such file or directory

CoretechR, 2023-07-23:
The entire obj folder inside OMOTE\LVGL Simulator\LvglPlatform\lvgl\examples\widgets is missing. 
It's there on my machine but GitHub for Desktop does not recognize it so it was never committed. 
There is no gitignore file that would make GitHub ignore this folder. Do you have any idea what would cause this?

MatthewColvin, 2023-07-23:
Yeah this is a sub module inside the other repo can you tell me the commit SHA of the LVGL Simulator you are using? 

I can fork it and then pull it into the OMOTE repo as a submodule.

CoretechR, 2023-07-23:
It took some convincing but the missing files are online now (53f9d1b). 
I will try and switch to a proper submodule.

> I am unsure how those interact with one another at this point but I look forward to digging in.

The simulator and the Platformio projects don't have any direct connection at the moment. I just carefully copy the code between them. It's not very elegant, but it works. There are just too many differences to make these 1:1 compatible.

MatthewColvin, 2023-07-23:
thanks for pulling in those files. This is so sweet!!! 

> The simulator and the Platformio projects don't have any direct connection at the moment. I just carefully copy the code between them. It's not very elegant, but it works. There are just too many differences to make these 1:1 compatible.

Is there something you would like me to try and implement? or I can look into the easiest way to share code between the sim and the HW?

closed 2023-07-23 by MatthewColvin

CoretechR, 2023-07-23:
If you want to, sure. Linking the simulator to the embedded code would be nice, but its not really crucial for the further development. 
I don't have any concrete plans on what to do next with this project, but I'd like to make the existing functionality as stable as possible for the people that are building their boards. After that, the next big step would be adding some kind of customization to the UI.

Dark1886, 2023-07-28:
There are other projects that use Platformio to simulate the LVGL screen, although I'm not sure if that really allows for editing of the display in any easy way. I'm intending to look a theses when I have some time, but I haven't had as much time as I'd like recently. I admit I'm also not familiar with LVGL so maybe others would have more information. 

Information collection:
https://blog.drorgluska.com/2022/07/lvgl-esp32-and-desktop-development.html
https://github.com/lvgl/lv_platformio