OMOTE-Community/OMOTE-Firmware #32: R21 seems to have wrong value

Salecorp, 2023-08-10:
It seems like 3R3 is too low for protecting the IR LED.
Is this a mistake or on purpose?

CoretechR, 2023-08-10:
The small resistor is an intentional choice to get the best possible range out of the LED. Most 5mm IR LEDs are specified for a pulse current of around 1A. The LED is controlled by the IR library so it should be fine. But if a programming error occurs and the LED is powered on for more than a few milliseconds, it will get damaged. Some kind of protection circuit would be part of the next hardware revision.

Salecorp, 2023-08-10:
I see, thank you for the clarification!

samdowrick, 2023-08-10:
@CoretechR 
Here's an approach that utilizes the 555 timer for protection:

1. Connect +3.3V Supply to 555 Timer:
        Connect the +3.3V supply to the power input of the 555 timer.
2.Configure 555 Timer as One-Shot:
        Set up the 555 timer in monostable mode to generate a pulse when triggered.
3.Trigger 555 Timer:
        Connect the trigger input of the 555 timer to the microcontroller's output. The microcontroller will trigger the 555 timer to 
        start its timing cycle.
5. Connect 555 Timer Output to Transistor Base:
        Connect the output of the 555 timer to the base of an NPN transistor (e.g., 2N2222 or similar).
6. Connect Transistor Collector to +3.3V:
        Connect the collector of the transistor to the +3.3V supply.
7. Connect Transistor Emitter to Q5 (DMG2301L):
        Connect the emitter of the transistor to the source pin (pin2) of the DMG2301L mosfet.
Rest of the circuit remains unchanged.

I can modify the schematic and upload an image to show this if it doesn't make sense.

CoretechR, 2023-08-11:
@samdowrick Your idea makes perfect sense. I was actually considering using a resettable fuse or a big capacitor to limit the energy to the LED. But yours is a much more clever solution. Thank you!
I've added it to the list of suggestions for a new hardware revision: https://github.com/CoretechR/OMOTE/discussions/24

CoretechR, 2023-08-16:
Fixed in commit #33

closed 2023-08-16 by CoretechR