OMOTE-Community/OMOTE-Firmware #36: There is a certain probability of restarting ESP32 several times during startup

vihor3, 2023-09-09:
`[    15][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   440][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=19 scl=22 freq=400000
[   441][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[   568][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY

Brownout detector was triggered`
The UART output is like this, sometimes this segment will repeat 4-5 times, sometimes 1-2 times, and then the screen will light up normally

MiniJe, 2023-09-09:
This is a power problem. You need a more capable power supply to feed the
ESP32. If using a USB-powered dev board, try replacing the USB cable and
connecting it directly to a USB port on your computer. A powered USB hub
might also help.
Also, you can add   WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable
brownout detector

void printBootReason()
{
    esp_reset_reason_t reset_reason = esp_reset_reason();

    switch (reset_reason)
    {
    case ESP_RST_UNKNOWN:    Serial.println("Reset reason can not be
determined"); break;
    case ESP_RST_POWERON:    Serial.println("Reset due to power-on
event"); break;
    case ESP_RST_EXT:        Serial.println("Reset by external pin
(not applicable for ESP32)"); break;
    case ESP_RST_SW:         Serial.println("Software reset via
esp_restart"); break;
    case ESP_RST_PANIC:      Serial.println("Software reset due to
exception/panic"); break;
    case ESP_RST_INT_WDT:    Serial.println("Reset (software or
hardware) due to interrupt watchdog"); break;
    case ESP_RST_TASK_WDT:   Serial.println("Reset due to task
watchdog"); break;
    case ESP_RST_WDT:        Serial.println("Reset due to other
watchdogs"); break;
    case ESP_RST_DEEPSLEEP:  Serial.println("Reset after exiting deep
sleep mode"); break;
    case ESP_RST_BROWNOUT:   Serial.println("Brownout reset (software
or hardware)"); break;
    case ESP_RST_SDIO:       Serial.println("Reset over SDIO"); break;
    }

    if (reset_reason == ESP_RST_DEEPSLEEP)
    {
        esp_sleep_wakeup_cause_t wakeup_reason = esp_sleep_get_wakeup_cause();

        switch(wakeup_reason)
        {
            case ESP_SLEEP_WAKEUP_UNDEFINED:    Serial.println("In
case of deep sleep: reset was not caused by exit from deep sleep");
break;
            case ESP_SLEEP_WAKEUP_ALL:          Serial.println("Not a
wakeup cause: used to disable all wakeup sources with
esp_sleep_disable_wakeup_source"); break;
            case ESP_SLEEP_WAKEUP_EXT0:         Serial.println("Wakeup
caused by external signal using RTC_IO"); break;
            case ESP_SLEEP_WAKEUP_EXT1:         Serial.println("Wakeup
caused by external signal using RTC_CNTL"); break;
            case ESP_SLEEP_WAKEUP_TIMER:        Serial.println("Wakeup
caused by timer"); break;
            case ESP_SLEEP_WAKEUP_TOUCHPAD:     Serial.println("Wakeup
caused by touchpad"); break;
            case ESP_SLEEP_WAKEUP_ULP:          Serial.println("Wakeup
caused by ULP program"); break;
            case ESP_SLEEP_WAKEUP_GPIO:         Serial.println("Wakeup
caused by GPIO (light sleep only)"); break;
            case ESP_SLEEP_WAKEUP_UART:         Serial.println("Wakeup
caused by UART (light sleep only)"); break;
        }
    }
}

vihor3, 2023-09-09:
Thank you. I have redrawn the PCB for this project and used HT7833's LDO, which may not be able to handle situations where WiFi and screen backlight are turned on simultaneously. I am going to try replacing the ME6211 for testing. This LDO has better dynamic response, but its static power consumption is a bit high. Perhaps it is also possible to try increasing the capacitance of the input and output capacitors to solve the problem

MiniJe, 2023-09-09:
Suggestion for measuring battery level:
How to measure battery level with ESP32 microcontroller | Andrey Ovcharov
<https://en.ovcharov.me/2020/02/29/how-to-measure-battery-level-with-esp32-microcontroller/>

All the best
Pau

vihor3, 2023-09-09:
For this project, ESP32-Pico-D4 no longer has more IO to use, so the functions in the article cannot be implemented unless ESP32-S3 is replaced

MiniJe, 2023-09-09:
you can use LCD_EN ...
the function for measuring the battery level should only be activated when
the remote control is at rest (placed) and the LCD is with the backlight off

CoretechR, 2023-09-09:
Is there a problem with the existing battery measurement? It's only a simple voltage divider but it works for me and has a very low power draw.

MiniJe, 2023-09-09:
Unfortunately the version with the divider is not stable. plus it
discharges the battery faster because the divider acts as a load.

There are many issues discussed on the forums, related to measuring the
level in the battery.
https://esp32.com/viewtopic.php?t=29582
https://forum.arduino.cc/t/fixing-the-non-linear-adc-of-an-esp32/699190
.. and other examples.
I use: https://eu.mouser.com/datasheet/2/609/MAX17300_MAX17313-3127492.pdf
In addition, your project is 5 stars. The above observations are from
personal experience and are intended to increase operational reliability

CoretechR, 2023-09-09:
The ADC of the ESP32 is notoriously bad and non-linear. The 600kOhm voltage divider uses 6uA, so it can give a rough estimate of the battery state. A fuel gauge like you suggested, is a much nicer solution. I would definitely consider this for a new hardware revision.

vihor3, 2023-09-09:
It is said that the ADC input impedance of ESP32 is less than 100K, so it is best to use a divider resistor that is less than 100K. However, reducing the resistance value will lead to an increase in power consumption. My solution for this is to use a P-MOS to control the BATT access to the voltage divider network. Of course, the IO of ESP32-Pico-D4 is currently insufficient, and the FLASH storage footprint seems to be full. Therefore, I suggest replacing it with a more powerful ESP32-S3 and supporting a larger FLASH IC

vihor3, 2023-09-09:
Of course, such changes will incur additional costs. If there are no special requirements, the current hardware is already sufficient to use, and the next work only needs to optimize the software

closed 2023-09-10 by vihor3