OMOTE-Community/OMOTE-Firmware #38: MQTT Support on Abstraction Branch

MatthewColvin, 2023-09-11:
When abstracting out wifi and other HW component MQTT support dropped off so need to add it back.

baerrs, 2024-03-04:
I've got the MQTT working (not the Keyboard).  

Before this, it would cause the WiFi to disconnect after the first MQTT publish. Sometimes the WiFi would recover, sometimes it wouldn't.  
Looking at the MQTT Server logs, it appeared that OMOTE was connecting as OMOTE, and then multiple different ClientID's.  

These changes stopped that.  
I had to disable the Keyboard.  I commented out all entries in 
I don't know if I had to do all this.. but it is working after i did.  
```
void register_device_keyboard_mqtt() {
  // commands[KEYBOARD_MQTT_UP]                   = makeCommandData(MQTT, {"esp32_keyboard_firetv/cmnd/UP",                  "PRESS"});
  // commands[KEYBOARD_MQTT_DOWN]                 = makeCommandData(MQTT, {"esp32_keyboard_firetv/cmnd/DOWN",                "PRESS"});
  // commands[KEYBOARD_MQTT_RIGHT]                = makeCommandData(MQTT, {"esp32_keyboard_firetv/cmnd/RIGHT",               "PRESS"});
  // commands[KEYBOARD_MQTT_LEFT]                 = makeCommandData(MQTT, {"esp32_keyboard_firetv/cmnd/LEFT",                "PRESS"});
  // commands[KEYBOARD_MQTT_SELECT]               = makeCommandData(MQTT, {"esp32_keyboard_firetv/cmnd/SELECT",              "PRESS"});
  // commands[KEYBOARD_MQTT_SENDSTRING]           = makeCommandData(MQTT, {"esp32_keyboard_firetv/cmnd/SENDSTRING"                  }); // payload must be set when calling commandHandler
  // commands[KEYBOARD_MQTT_BACK]                 = makeCommandData(MQTT, {"esp32_keyboard_firetv/cmnd/BACK",                "PRESS"});
  // commands[KEYBOARD_MQTT_HOME]                 = makeCommandData(MQTT, {"esp32_keyboard_firetv/cmnd/HOME",                "PRESS"});
  // commands[KEYBOARD_MQTT_MENU]                 = makeCommandData(MQTT, {"esp32_keyboard_firetv/cmnd/MENU",                "PRESS"});
  // commands[KEYBOARD_MQTT_SCAN_PREVIOUS_TRACK]  = makeCommandData(MQTT, {"esp32_keyboard_firetv/cmnd/SCAN_PREVIOUS_TRACK", "PRESS"});
  // commands[KEYBOARD_MQTT_REWIND_LONG]          = makeCommandData(MQTT, {"esp32_keyboard_firetv/cmnd/REWIND_LONG",         "PRESS"});
  // commands[KEYBOARD_MQTT_REWIND]               = makeCommandData(MQTT, {"esp32_keyboard_firetv/cmnd/REWIND",              "PRESS"});
  // commands[KEYBOARD_MQTT_PLAYPAUSE]            = makeCommandData(MQTT, {"esp32_keyboard_firetv/cmnd/PLAYPAUSE",           "PRESS"});
  // commands[KEYBOARD_MQTT_FASTFORWARD]          = makeCommandData(MQTT, {"esp32_keyboard_firetv/cmnd/FASTFORWARD",         "PRESS"});
  // commands[KEYBOARD_MQTT_FASTFORWARD_LONG]     = makeCommandData(MQTT, {"esp32_keyboard_firetv/cmnd/FASTFORWARD_LONG",    "PRESS"});
  // commands[KEYBOARD_MQTT_SCAN_NEXT_TRACK]      = makeCommandData(MQTT, {"esp32_keyboard_firetv/cmnd/SCAN_NEXT_TRACK",     "PRESS"});
  // commands[KEYBOARD_MQTT_MUTE]                 = makeCommandData(MQTT, {"esp32_keyboard_firetv/cmnd/MUTE",                "PRESS"});
  // commands[KEYBOARD_MQTT_VOLUME_INCREMENT]     = makeCommandData(MQTT, {"esp32_keyboard_firetv/cmnd/VOLUME_INCREMENT",    "PRESS"});
  // commands[KEYBOARD_MQTT_VOLUME_DECREMENT]     = makeCommandData(MQTT, {"esp32_keyboard_firetv/cmnd/VOLUME_DECREMENT",    "PRESS"});
}
```
I also made f few changes to hardware/mqtt.cpp
From:
```
PubSubClient mqttClient(espClient);
```
To:
```
PubSubClient mqttClient(MQTT_SERVER, MQTT_SERVER_PORT, espClient);

```
I also added     mqttClient.setKeepAlive(300); to WiFiEvent
This maybe specific to my MQTT server
```
// WiFi status event
void WiFiEvent(WiFiEvent_t event){
  //Serial.printf("[WiFi-event] event: %d\r\n", event);
  if(event == ARDUINO_EVENT_WIFI_STA_GOT_IP){
    // connection to MQTT server will be done in checkMQTTconnection()
    // mqttClient.setServer(MQTT_SERVER, 1883); // MQTT initialization
    // Set the MQTT keepalive to 300 seconds
    mqttClient.setKeepAlive(300);
    mqttClient.connect("OMOTE"); // Connect using a client id

  }
```

KlausMu, 2024-03-04:
Hi @baerrs this is the wrong issue to discuss it. This issue is about the abstraction branch, which is a [complete different software](https://github.com/CoretechR/OMOTE/tree/abstraction) 

It's interesting that with these changes it works for you. I would like to track it down to the real reason.
My first guess is still lack of memory. Did you try the latest version of my software, having a big improvement in memory usage? https://github.com/CoretechR/OMOTE/discussions/58 See latest post at bottom.

1. The comments on `commands[KEYBOARD_MQTT_UP]` are not necessary, you can revert them. These are only commands that get registered. As long as you don't use them in code, the MQTT messages will not be sent
2. with your change in the code, the connection `mqttClient.connect()` is now done at two places in the code. Without having a closer look, I think it is not really bad, but on the other hand no really necessary. Also setting the mqtt server and port like you did is not necessary, since this is already done at a different place.
3. The only substantial change seems to be the `mqttClient.setKeepAlive(300)`

What you could do is:
- try the latest version which is linked above and see if wifi issues are gone
- or, if you understand the memory issue, stay in the current version, reduce lvgl static heap to 32k, use as less guis as possible and see if wifi problems are gone
- revert all your changes, keep only `setKeepAlive`, and retry-

baerrs, 2024-03-04:
@KlausMu  Thank you.  
I switched to your forked abstract code.  It works without changing anything.

KlausMu, 2024-03-04:
Ah, that's good to hear. It was always my guess that all WiFi problems are only related to memory.
And to be precise, because my code is not a fork of the "abstract branch"

main branch
https://github.com/CoretechR/OMOTE/tree/main
latest stable version of my software
not abstract, not using c++ classes, and not being object oriented 

latest test version
https://github.com/KlausMu/OMOTE/tree/modular-approach
currently including memory optimizations, which is not yet in the main branch

abstract version
https://github.com/CoretechR/OMOTE/tree/abstraction
a completely different software from @MatthewColvin , including classes and object oriented code for hardware abstraction
currently not being further developed

KlausMu, 2024-03-19:
I think we can close this issue, since MQTT Support is now available in the new simulator.
@MatthewColvin do you agree?

MatthewColvin, 2024-05-16:
Yeah makes sense to me :)

closed 2024-05-16 by MatthewColvin