OMOTE-Community/OMOTE-Firmware #47: Bluetooth Keyboard and mouse mode

MatthewColvin, 2023-10-06:
Make the remote able to advertise and connect to HID hosts to allow keyboard and mouse control of devices like android tv.

Wildpig954, 2023-10-06:
This function would be great. My Harmony hub has keyboard and when I change between TV, windows and nvidia shield the keyboard switches too. This would be a great option to have.

phant0mias, 2023-10-06:
Hi,
that is not difficult. I already tried it.

you can use:
https://github.com/T-vK/ESP32-BLE-Keyboard
or 
https://github.com/lemmingDev/ESP32-BLE-Gamepad

With the gamepad library I managed to use the remote control with the Amazon Fire Stick. I can write this weekend what commands from the gamepad are used to control the fire stick.

When you add these libraries you should use the newest platformio.ini file to get more flash memory (https://github.com/CoretechR/OMOTE/pull/43)

A problem is that the boot time is getting higher

phant0mias, 2023-10-08:
For controlling a amazon fire stick I use the bleGamepad class. After connecting the device in the fire stick settings as gamecontroller you can send the remote commands.
Automatic reconnect after power down the TV and remote is working too.

You have to experiment with it and check the examples of this class.

If somebody found some additional information like how to change the channel and volume with bluetooth it would be great.

I had an effort with the followings commands:

#include <BleGamepad.h>

BleGamepad m_bleGamepad

void setup()
{
..... 
    BleGamepadConfiguration bleGamepadConfig;
    bleGamepadConfig.setAutoReport(true);
    bleGamepadConfig.setControllerType(CONTROLLER_TYPE_GAMEPAD);
    bleGamepadConfig.setIncludeStart(true);
    bleGamepadConfig.setIncludeSelect(true);
    bleGamepadConfig.setIncludeMenu(true);
    bleGamepadConfig.setIncludeHome(true);
    bleGamepadConfig.setIncludeBack(true);

    bleGamepad.begin(&bleGamepadConfig);
....
}

as commands you can send:

// Back
bleGamepad.pressBack();
delay( 5 );
bleGamepad.releaseBack();

// Down
bleGamepad.setHat1(HAT_DOWN);
delay( 5 );
bleGamepad.setHat1(HAT_CENTERED);

// Home
bleGamepad.press(BUTTON_12);
delay( 5 );
bleGamepad.release(BUTTON_12);

// Left
bleGamepad.setHat1(HAT_LEFT);
delay( 5 );
bleGamepad.setHat1(HAT_CENTERED);

// Menu
bleGamepad.press(BUTTON_11);
delay( 5 );
bleGamepad.release(BUTTON_11);

// OK
bleGamepad.press(BUTTON_1);
delay( 5 );
bleGamepad.release(BUTTON_1);

// Right
bleGamepad.setHat1(HAT_RIGHT);
delay( 5 );
m_bleGamepad.setHat1(HAT_CENTERED);

// Up
m_bleGamepad.setHat1(HAT_UP);
delay( 5 );
m_bleGamepad.setHat1(HAT_CENTERED);

phant0mias, 2023-10-09:
I opened a discussion for this theme

https://github.com/CoretechR/OMOTE/discussions/48

KlausMu, 2024-03-13:
Issue closed, because in main BLE keyboard support is available. FireTV can be controller, others as well.

closed 2024-03-13 by KlausMu