OMOTE-Community/OMOTE-Firmware #49: Case manufacturing like https://www.wiphone.io/

devenator, 2023-10-15:


devenator, 2023-10-15:
Is a Case manufacturing like https://www.wiphone.io (basically an aluminum profile with two plates, eighter also aluminum or plexiglass) feasible? Does anyone know a company that does somethin kline that...?

GordonZed, 2024-04-22:
The wiphone case doesn't look to me (at least from the hero image on the homepage) like just two aluminum plates; it looks like aluminum that's been stamped (die cut *and* bent/folded), so while anything is feasible if you're willing to pay for it, it's far more feasible if you design something that requires only cutting, because any service that provides CNC milling or laser cutting services will be able to do that for far less expense.

I would look at something like [ponoko.com](https://ponoko.com), although I haven't looked into them in quite a while and they seem to have gone from a simple laser cutting service for hobbyists to targeting small tech startups, but regardless, a quote is free, so the worst that can happen is the quote is more than you want to spend, and you either keep looking, or consider cheaper materials. I'd also look at [Shapeways](https://www.shapeways.com), who are known for 3D printing, but they provide a wide range of manufacturing and prototyping services, and again, a quote is free, and every quote that hurts is an educational experience.

What I would do is check local marketplaces (Craigslist, Kijiji [this might be Canada only], FB Mktplce, etc.) and see if you can find someone with a hobby shop who provides custom laser-cutting/CNC milling as a side hustle.

In the end, all of this is available, but depending on the desired materials/processes, can be prohibitively expensive. Your best bet would be laser cut acrylic, and maybe a thin veneer of brushed aluminum on the exterior faces for the final look. It's very cheap to get CNC cut acrylic parts, but aluminum beyond, say a sixteenth or two is probably going to increase in price rapidly as the material thickness goes up.

closed 2025-04-06 by KlausMu