OMOTE-Community/OMOTE-Firmware #55: Battery charge status not reliable

KlausMu, 2024-01-13:
Hi Maximilian,

Great project, thanks a lot for it!

On my board (created with your PCB files), "battery_ischarging = !digitalRead(CRG_STAT);" does not deliver reliable results.
Sometimes result is correct, sometimes not. Wrong detection happens both when battery is empty or when it is full.
Seems to be a little bit random.

Could it be that you cannot both connect a LED to STAT and at the same time digitally read the state of STAT?
The MCP73831 datasheet states:
`STAT is an output for connection to an LED for charge status indication. **Alternatively**, a pull-up resistor can be
applied for interfacing to a host microcontroller.`
So maybe the digital read of STAT is not longer reliable if a LED is connected.
Unfortunately STAT is connected to GPIO21, which cannot read analog values.

What do you think?

CoretechR, 2024-01-13:
Hi Klaus, I have not heard about this problem before. The charging indication always seemed reliable enough to me. Do you have a multimeter to measure the voltage on the charging status pin?

KlausMu, 2024-01-14:
Ok. Battery is almost full, 4.1 V.

1. Charger connected:
LED on, Voltage on STAT is 0.11 V, !digitalRead(CRG_STAT) == true

2. Charger not connected:
LED off, Voltage on STAT is 1.12 V, !digitalRead(CRG_STAT) == true

CoretechR, 2024-01-14:
Thanks for checking this! Unfortunately I just found a problem with my circuit:
[image]
The charger is supplied by VBUS and I connected the status output directly to the ESP32. 
That's bad in two ways: 
1. It's a push-pull output so when charging is complete, the charger sets the status pin to 5V. The clamping diodes in the ESP32 will protect the controller to some degree, but that's not something to rely on. 
2. When USB is disconnected, there is no pull-up voltage to set the status to a defined level, which might explain your issue.

I'm really sorry that I overlooked this. I even used a similar circuit in prior projects, somehow without any problems.

As for the ESP32 getting damaged, I think we should be OK given the number of working boards in circulation. To fix your unreliable status indicator, the LED could be disconnected from VBUS and wired to +3.3V. In any case I will create a REV3 to fix this and other minor issues.

KlausMu, 2024-01-14:
Ok, great, that you found this issue. I also think 5V on the ESP32 input shouldn't be a real problem, I also had this before in other projects.

If you want to, you could think about a circuit which recognizes all three states. Currently HIGH-Z cannot be recognized. But it is only nice to have ... https://electrical.codidact.com/posts/286209

CoretechR, 2024-01-14:
Thank you for pointing me to this problem in the first place! And sure, if I do a new PCB revision, I will also look into a proper detection circuit.

closed 2024-06-17 by CoretechR