OMOTE-Community/OMOTE-Firmware #65: Discord Link corrent?

bjoerns1983, 2024-02-22:
Hi there,
is the Discord Link in the readme still correct?

KlausMu, 2024-02-22:
Yes, for me it works. https://discord.com/channels/1138116475559882852/1153343867681243279

aristeu, 2024-02-24:
Link doesn't work for me either. Wondering if the server requires invite. Someone could create an invite (down arrow on the server icon, invite people, edit invite link, set it to not expire) and replace that link in the documentation.

CoretechR, 2024-02-24:
That is strange. Discord shows me that the link in the readme is active and set to never expire. 
[image]

aristeu, 2024-02-24:
The format for invite URLs are

```
https://discord.gg/<10 characters>
```

while the link looks like something it'd be copied from the URL bar while browsing the server/channel.
This is what shows when clicking on the link (in a browser I'm authenticated at discord):

[image]

You might be able to reproduce the issue by logging off your discord account (or using a different browser you're not authenticated).

CoretechR, 2024-02-24:
Are we talking about this link?
https://discord.gg/5PnYFAsKsG

Because it works for me. I also tried it in incognito mode. Or what other link is there in the readme?

aristeu, 2024-02-24:
Ah, I see what's happening. I clicked on the Discord link to buy the PCB channel:

```
Check out the buy-sell page on the [Discord](https://discord.com/channels/1138116475559882852/1153343867681243279) to see if you can share the cost of the PCBs and components with others.
```

And didn't get to the end of the page so I didn't see the invite. Invite link works, thanks.

Dark1886, 2024-02-25:
@CoretechR otherwise use this link in the readme 

https://discord.gg/cTR5d8AYFj

closed 2024-02-26 by bjoerns1983