OMOTE-Community/OMOTE-Firmware #83: PCB Revision 4

osmanates, 2024-06-11:
Hi,

I saw in the schematic some components like C27, U10 is marked with X. Are these not supported to be in the pcb? BOM.csv file seems to not have these but Kicad schematic and pcb files seems to have these components. I plan to get PCBA service from JLCPCB so İ will need to upload the gerber files, bom and cpl. Not sure what files to use since those have these components Any help here is appreciated.

KlausMu, 2024-06-13:
@CoretechR can you help?

CoretechR, 2024-06-13:
U10 is an optional reset generator for the display controller (see #70). U10 and C27 are marked as unpopulated in the schematic. 
If somehow these two components ended up in your PCBA order, you can remove them from the BOM. Or you can leave the in. Functionally it should not make a difference.

osmanates, 2024-06-13:
Thanks. There are also C18, C19 and J4. Same applies for them as well?

CoretechR, 2024-06-14:
Yes, those should not be soldered to the PCB by default.

closed 2024-06-17 by CoretechR